pfalcon/esp-open-sdk #329: [ERROR]  You must NOT be root to run crosstool-NG

kaizoku-619, 2018-05-31:
Hi, I'm trying to make the sdk but I'm facing the following error.
Even thought I created a user and add it to sudoers, when making, the problem remains.
___________________________________________________________________________________
Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/root/Development/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/root/Development/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/root/Development/esp-open-sdk/crosstool-NG'
[ERROR]  You must NOT be root to run crosstool-NG
[00:00] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 1
make[2]: Leaving directory '/root/Development/esp-open-sdk/crosstool-NG'
../Makefile:133: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/root/Development/esp-open-sdk/crosstool-NG'
Makefile:129: recipe for target '/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
___________________________________________________________________________________

pfalcon, 2018-07-10:
Well, "You must NOT be root to run crosstool-NG".

closed 2018-07-10 by pfalcon

cirosantilli, 2018-11-01:
There is also `CT_ALLOW_BUILD_AS_ROOT=y` if you really know what you are doing: https://stackoverflow.com/questions/17466017/i-got-such-an-error-when-ct-ng-build/53099177#53099177 e.g. within Docker.

RadoslavMarinov, 2019-02-11:
Well when I use just `make` it says: autom4te: can not open ......: Permission denied