pfalcon/esp-open-sdk #335: Cannot make build toolchain

MrKevinWeiss, 2018-06-26:
It seems like the link to get the binutils-2.25.1 is broken, is it possible to replace with something more stable?

We cannot download from the ftp site with or without proxy.

```
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180626.094025
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.05s (at 00:01)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[ERROR]    The 'binutils-2.25.1' sources were partially extracted.
[ERROR]    Please remove first:
[ERROR]     - the source dir for 'binutils-2.25.1', in '/home/kevinweiss/opt/esp-open-sdk/crosstool-NG/.build/src'
[ERROR]     - the file '/home/kevinweiss/opt/esp-open-sdk/crosstool-NG/.build/src/.binutils-2.25.1.extracting'
[ERROR]    I'll stop now to avoid any carnage...
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
[ERROR]  >>        called from: CT_Extract[scripts/functions@959]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:00.96)
[00:01] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 1
make[2]: Leaving directory '/home/kevinweiss/opt/esp-open-sdk/crosstool-NG'
../Makefile:133: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/kevinweiss/opt/esp-open-sdk/crosstool-NG'
Makefile:129: recipe for target '/home/kevinweiss/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/kevinweiss/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

```

jcmvbkbc, 2018-07-09:
> We cannot download from the ftp site with or without proxy.

But if you read the log carefully...

> [INFO ]  Extracting and patching toolchain components
> [ERROR]    The 'binutils-2.25.1' sources were partially extracted.
> [ERROR]    Please remove first:
> [ERROR]     - the source dir for [...]

...you see that is says that it couldn't extract binutils archive, because it already was partially extracted.

> We cannot download from the ftp site with or without proxy.

I have just checked wget https://ftp.gnu.org/gnu/binutils/binutils-2.25.1.tar.gz , worked perfectly for me.

pfalcon, 2018-07-10:
Thanks @jcmvbkbc for response. Also, looking at the linked issue (https://github.com/RIOT-OS/RIOT/pull/8921), the matter might have been resolved. Closing.

closed 2018-07-10 by pfalcon