pfalcon/esp-open-sdk #340: make[1]: *** crosstool-NG: No such file or directory.  Stop.

youngda, 2018-08-28:
#sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
    sed git unzip bash help2man wget bzip2 libtool-bin
#git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
#cd esp-open-sdk
#make STANDALONE=y
#export PATH=/home/youngda/esp/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
#cd ESP8266_NONOS_SDK-2.1.0-18-g61248df
#cp -r examples/IoT_Demo ./
#./gen_misc.sh
start...

../../Makefile:106: warning: overriding recipe for target 'clean'
../Makefile:317: warning: ignoring old recipe for target 'clean'
../../Makefile:184: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../../Makefile:179: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: *** crosstool-NG: No such file or directory.  Stop.
../../Makefile:143: recipe for target 'crosstool-NG/ct-ng' failed
make: *** [crosstool-NG/ct-ng] Error 2

`I don't know what happened.`

youngda, 2018-08-30:
cd esp/esp-open-sdk/
mv Makefile Makefile_backup
sudo rm -r ESP8266_NONOS_SDK-2.1.0-18-g61248df
Download the new SDK and you're ready to use it.

sifourquier, 2019-01-13:
similar for me.
and try with ESP8266_NONOS_SDK-3.0 extracted in esp-open-sdk and similar issue.

sifourquier, 2019-01-13:
i think i have found a fix
go to your ESP8266_NONOS_SDK
open Makefile
comment 2 last line
#PDIR := ../$(PDIR)
#sinclude $(PDIR)Makefile

AndreRABR, 2020-09-18:
Tried this and now I am getting this issue.

[image]