pfalcon/esp-open-sdk #341: Building toolchain fails on ARM machine

igagis, 2018-09-18:
I'm trying to build the sdk on my ARM machine running Debian stretch.

```
$ uname -a
Linux u3 3.8.13.30 #1 SMP PREEMPT Wed Oct 26 17:53:58 CEST 2016 armv7l GNU/Linux
```

After I run
```
make STANDALONE=y
```

it starts doing something but at some point it fails with this error message:
```
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing GMP for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gmp_backend[scripts/build/companion_libs/100-gmp.sh@106]
[ERROR]  >>        called from: do_gmp_for_host[scripts/build/companion_libs/100-gmp.sh@62]
[ERROR]  >>        called from: do_companion_libs_for_host[scripts/build/companion_libs.sh@36]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
```
I looked through the `Known Isuues` document but could not find anything similar.

Here is the `build.log`:
[build.log](https://github.com/pfalcon/esp-open-sdk/files/2394630/build.log)

igagis, 2018-09-18:
I found similar discussion about building `GMP`:
https://gmplib.org/list-archives/gmp-bugs/2013-July/003095.html

and there they suggest to remove the `-mthumb` compilation flag

jeremyd2019, 2018-09-19:
You need a patch for gmp:
https://gmplib.org/repo/gmp/raw-rev/676e2d0f0e4d

You can see how to put it into crosstool-NG here:
https://github.com/jeremyd2019/crosstool-NG/commit/f37aa73ab1eab17999392c563b71099e5839ee23

igagis, 2018-09-19:
@jeremyd2019 Thanks, I will try it later today. Did you submit that patch to `crosstol-NG` upstream?

igagis, 2018-09-19:
Thanks, it worked!

igagis, 2018-09-19:
Another error followed:

```
[INFO ]  Installing pass-2 core C gcc compiler
[ERROR]    collect2: error: ld returned 1 exit status
[ERROR]    make[3]: *** [build/genhooks] Error 1
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@514]
[ERROR]  >>        called from: do_gcc_core_pass_2[scripts/build/cc/100-gcc.sh@160]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc.sh@42]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
```

Build log:
[build.log](https://github.com/pfalcon/esp-open-sdk/files/2398209/build.log)

igagis, 2018-09-30:
On raspberry pi I was able to build it successfully.

closed 2018-09-30 by igagis