pfalcon/esp-open-sdk #343: Build error at first time, cannot extract component

AbuAhmed99, 2018-10-27:
Hello all,

i'm on Ubuntu 18.04 , I tried following the instructions

1. $ sudo apt-get install libtool-bin
2. $ git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
3. $ cd esp-open-sdk
4. $ make STANDALONE=y 

when building it, i cannot extract toolchain component and i get the errors:

Makefile:184: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:179: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory '/home/praz/esp-open-sdk/crosstool-NG'
../Makefile:184: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:179: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... no
checking for objcopy... objcopy
checking for absolute path to objcopy... /usr/bin/objcopy
checking for gobjdump... no
checking for objdump... objdump
checking for absolute path to objdump... /usr/bin/objdump
checking for greadelf... no
checking for readelf... readelf
checking for absolute path to readelf... /usr/bin/readelf
checking for gperf... gperf
checking for absolute path to gperf... /usr/bin/gperf
checking for bison... bison
checking for flex... flex
checking for makeinfo... makeinfo
checking for cut... cut
checking for stat... stat
checking for readlink... readlink
checking for wget... wget
checking for tar... tar
checking for gzip... gzip
checking for bzip2... bzip2
checking for help2man... help2man
checking for gpatch... no
checking for patch... patch
checking for absolute path to patch... /usr/bin/patch
checking for bash >= 3.1... /bin/bash
checking for GNU awk... /usr/bin/gawk
checking for GNU make >= 3.80... /usr/bin/make
checking whether /usr/bin/make sets $(MAKE)... yes
[... 159 lines cut ...]
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Extract[scripts/functions@982]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 22:44.37)
[22:45] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/praz/esp-open-sdk/crosstool-NG'
../Makefile:133: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/praz/esp-open-sdk/crosstool-NG'
Makefile:129: recipe for target '/home/praz/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/praz/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

Thanks in advance and hope anyone can helps me.

ha-yi, 2018-11-07:
I am having the same problem, here is the last lines of build.log in `crosstool-NG/build.log` file.

```
[ALL  ]    Overiding config.guess and config.sub
[ALL  ]    /Volumes/case-sensitive/esp-open-sdk/crosstool-NG/scripts/config.guess -> ./config.guess
[ALL  ]    /Volumes/case-sensitive/esp-open-sdk/crosstool-NG/scripts/config.sub -> ./config.sub
[DEBUG]    ==> Executing: 'touch' '/Volumes/case-sensitive/esp-open-sdk/crosstool-NG/.build/src/.ncurses-6.0.patched'
[DEBUG]    ==> Executing: 'rm' '-f' '/Volumes/case-sensitive/esp-open-sdk/crosstool-NG/.build/src/.ncurses-6.0.patching'
[DEBUG]    ==> Executing: 'touch' '/Volumes/case-sensitive/esp-open-sdk/crosstool-NG/.build/src/.binutils-2.25.1.extracting'
[DEBUG]    Entering '/Volumes/case-sensitive/esp-open-sdk/crosstool-NG/.build/src'
[EXTRA]    Extracting 'binutils-2.25.1'
[DEBUG]    ==> Executing: 'mkdir' '-p' 'binutils-2.25.1'
[DEBUG]    ==> Executing: 'tar' '--strip-components=1' '-C' 'binutils-2.25.1' '-xv' '-f' '-'
[FILE ]    tar: Unrecognized archive format
[FILE ]    tar: Error exit delayed from previous errors.
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: CT_Extract[scripts/functions@983]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 0:11.00)
```

WTPGaming, 2020-05-31:
Same issue!