pfalcon/esp-open-sdk #345: ESP8266_NONOS_SDK-3.0  Buil Fail

pacmac, 2018-11-13:
I just updated to ESP8266_NONOS_SDK-3.0 using:

```
  git fetch origin pull/344/head
  git commit
  git checkout -b pullrequest FETCH_HEAD
  git submodule update --init
  make clean-sdk clean-sysroot all
```

Which I hope is correct, however now my build fails with:

```
.....
CC ../../py/mpprint.c
In file included from ./esp_mphal.h:76:0,
                 from ../../py/mphal.h:32,
                 from ../../py/mpprint.c:33:
./etshal.h:9:6: error: conflicting types for 'ets_delay_us'
 void ets_delay_us(uint16_t us);
      ^
In file included from ./esp_mphal.h:69:0,
                 from ../../py/mphal.h:32,
                 from ../../py/mpprint.c:33:
/nfs/qnap/data/uPython/build/micropython/ports/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint32_t us);
      ^
../../py/mkrules.mk:47: recipe for target 'build/py/mpprint.o' failed
make: *** [build/py/mpprint.o] Error 1
```

Any suggestions on what the cause may be ?

spaivaras, 2018-11-19:
I actually don't see any reference to etc_delay_* in espressif reference guide at all https://www.espressif.com/sites/default/files/documentation/2C-ESP8266_Non_OS_SDK_API_Reference__EN.pdf 
So errors seems logical, if they remove these

hjiawei, 2018-11-20:
I think this might be related to https://github.com/espressif/ESP8266_NONOS_SDK/commit/4671b17cc9fc6ed6787c2d310daf8accccf29c8d, which is merged into nonos sdk v2.2.0+.

hjiawei, 2018-11-20:
@pacmac Just FYI. I have built esp-open-sdk with esp8266 nonos sdk 2.2.1 and here is the change I need to make in micropython to get a successful build. I just got a build but haven't loaded it to my device.

```diff
diff --git a/ports/esp8266/etshal.h b/ports/esp8266/etshal.h
index 8d6457311..013160679 100644
--- a/ports/esp8266/etshal.h
+++ b/ports/esp8266/etshal.h
@@ -6,7 +6,7 @@
 // see http://esp8266-re.foogod.com/wiki/Random_Number_Generator
 #define WDEV_HWRNG ((volatile uint32_t*)0x3ff20e44)
 
-void ets_delay_us(uint16_t us);
+void ets_delay_us(uint32_t us);
 void ets_intr_lock(void);
 void ets_intr_unlock(void);
 void ets_isr_mask(uint32_t mask);
```

jackinto, 2019-04-01:
Hi Any luck building version 3.0?
I am also stuck trying to build it...
Thanks