pfalcon/esp-open-sdk #350: missing path in MacOS build instruction

bowerymarc, 2018-11-23:
In Readme.md, the macOS build instructions yields the error (Sierra):

> checking for gobjcopy... no
> checking for objcopy... no
> configure: error: missing required tool: gobjcopy objcopy
> make[1]: *** [_ct-ng] Error 1
> make: *** [crosstool-NG/ct-ng] Error 2

Adding the following to the path fixes it:

`export PATH="/usr/local/opt/binutils/bin:$PATH"`

toremielck, 2018-12-18:
> In Readme.md, the macOS build instructions yields the error (Sierra):
>
>
> Adding the following to the path fixes it:
>
> `export PATH="/usr/local/opt/binutils/bin:$PATH"`

This fixed it for me as well. Thanky you so much!

baaskaas, 2018-12-18:
> In Readme.md, the macOS build instructions yields the error (Sierra):
>
>
> Adding the following to the path fixes it:
>
> `export PATH="/usr/local/opt/binutils/bin:$PATH"`

Did not fix it for me, I did this right before I build it using this guide:

https://github.com/OtakuDigitChoc/esp-homekit-demo-stepByStep

I had the struggle on my Mac mini 2012 Server 
and my MBA 2013. Both on 10.14.2
MBA was a clean install. 

What I did was before I build it, I paste the export rule in the terminal. Then I use the following build rule:
make toolchain esptool libhal STANDALONE=n

my terminal output at the end:
d: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [nconf] Error 1
make: *** [build-lib-kconfig] Error 2
make: *** [build] Error 2
make[1]: *** [_ct-ng] Error 2
make: *** [crosstool-NG/ct-ng] Error 2

daviddpd, 2018-12-23:
See my patches on #342

ialidigital, 2020-12-10:
> Did not fix it for me, I did this right before I build it using this guide:
>
> https://github.com/OtakuDigitChoc/esp-homekit-demo-stepByStep
>
> I had the struggle on my Mac mini 2012 Server
> and my MBA [...]

I had this same issue, 

root was:
ld: warning: directory not found for option '-L/usr/local/Cellar/ncurses/6.1/lib'

I did this temp workaround (as couldn't get hold of 6.1 via brew):
sudo ln -s /usr/local/Cellar/ncurses/6.2/lib /usr/local/Cellar/ncurses/6.1