pfalcon/esp-open-sdk #356: Bash 5 compilation issue

Ergus, 2019-01-17:
Hi bash5 is out in distributions like arch linux, but the regex to check the version in crosstool-NG/configure.ac does not include it. This fixes the issue:

diff --git a/configure.ac b/configure.ac
index 5d512fe8..872a73de 100644
--- a/configure.ac
+++ b/configure.ac
@@ -190,7 +190,7 @@ AC_CACHE_VAL([ac_cv_path__BASH],
 AC_CACHE_CHECK([for bash >= 3.1], [ac_cv_path__BASH],
     [AC_PATH_PROGS_FEATURE_CHECK([_BASH], [bash],
         [[_BASH_ver=$($ac_path__BASH --version 2>&1 \
-                     |$EGREP '^GNU bash, version (3\.[1-9]|4)')
+                     |$EGREP '^GNU bash, version (3\.[1-9]|4|5)')
           test -n "$_BASH_ver" && ac_cv_path__BASH=$ac_path__BASH ac_path__BASH_found=:]],                                                           
         [AC_MSG_RESULT([no])
          AC_MSG_ERROR([could not find bash >= 3.1])])])

pfalcon, 2019-01-17:
crosstool-NG/configure.ac is part of crosstool-NG project, not this project.

closed 2019-01-17 by pfalcon

nashley, 2019-03-18:
Now that crosstool-NG [has been updated](https://github.com/jcmvbkbc/crosstool-NG/pull/56), can the submodule commit referencing it be updated to pull in the change?

rsnyman, 2019-03-22:
Yes, please, I was trying to figure out why it wasn't building for me.

milkpirate, 2022-01-03:
> Yes, please, I was trying to figure out why it wasn't building for me.

@pfalcon yes I agree with @rsnyman, could you please do so?

igagis, 2022-01-03:
@milkpirate see #387