pfalcon/esp-open-sdk #358: Issues building toolchain

hfosteriv, 2019-02-16:
I have made sure that the required components of the toolchain are installed on the machine via homebrew, but when the script gets to actually building the toolchain, it does not get past the ncurses installation. 

The output was: 

esp-open-sdk $ make toolchain esptool libhal STANDALONE=n
Makefile:186: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:181: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C crosstool-NG -f ../Makefile _toolchain
../Makefile:186: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:181: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./ct-ng xtensa-lx106-elf
gmake[2]: Entering directory '/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG'
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
gmake[2]: Leaving directory '/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
gmake[2]: Entering directory '/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190215.221936
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.00s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 0.00s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[ERROR]    clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ERROR]    gmake[4]: *** [Makefile:254: clear] Error 1
[ERROR]    gmake[3]: *** [Makefile:115: all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing ncurses for build'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: do_ncurses_backend[scripts/build/companion_libs/220-ncurses.sh@151]
[ERROR]  >>        called from: do_ncurses_for_build[scripts/build/companion_libs/220-ncurses.sh@35]
[ERROR]  >>        called from: do_companion_libs_for_build[scripts/build/companion_libs.sh@29]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:26.00)
[00:26] / gmake[2]: *** [ct-ng:152: build] Error 1
gmake[2]: Leaving directory '/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make: *** [crosstool-NG/.built] Error 2 

I also had another issue that required me downgrading my Xcode version to 9.2, and I also have a case sensitive .dmg mounted as instructed. 

Any help would be great!

hfosteriv, 2019-02-16:
For those interested in he build.log file, these are the last lines:

[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/infocmp.c -o ../objects/infocmp.o
[ALL  ]    echo "#define __TRANSFORM_H 1"					>>transform.h
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/clear.c -o ../objects/clear.o
[ALL  ]    echo "#include <progs.priv.h>"					>>transform.h
[ALL  ]    echo "extern bool same_program(const char *, const char *);"	>>transform.h
[ALL  ]    /bin/bash -c 'if test -n "" ; then echo "#define SUFFIX_IGNORED """>>transform.h; fi'
[ALL  ]    echo "#define PROG_CAPTOINFO "`echo captoinfo|   sed 's/$//'|sed 's,x,x,'|sed 's/$//'`""		>>transform.h
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    echo "#define PROG_INFOTOCAP "`echo infotocap|   sed 's/$//'|sed 's,x,x,'|sed 's/$//'`""		>>transform.h
[ALL  ]    echo "#define PROG_RESET     "`echo reset|       sed 's/$//'|sed 's,x,x,'|sed 's/$//'`""		>>transform.h
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/tabs.c -o ../objects/tabs.o
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/toe.c -o ../objects/toe.o
[ALL  ]    echo "#define PROG_INIT      "`echo init|        sed 's/$//'|sed 's,x,x,'|sed 's/$//'`""		>>transform.h
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    echo "#endif /* __TRANSFORM_H */"				>>transform.h
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/tic.c -o ../objects/tic.o
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -c /Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/transform.c -o ../objects/transform.o
[ALL  ]    clang: warning: argument unused during compilation: '--param max-inline-insns-single=1200' [-Wunused-command-line-argument]
[ALL  ]    x86_64-build_apple-darwin18.2.0-gcc ../objects/clear.o -L../lib -Wl,-search_paths_first -DHAVE_CONFIG_H -I. -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs -I../include -I/Volumes/case-sensitive/esp-homekit-demo-master/esp-open-sdk/crosstool-NG/.build/src/ncurses-6.0/progs/../include   -D_DARWIN_C_SOURCE -DNDEBUG -O2 -g  -no-cpp-precomp  --param max-inline-insns-single=1200 -lncurses    -o clear
[... cut ...]
[ERROR]  >>        called from: do_ncurses_for_build[scripts/build/companion_libs/220-ncurses.sh@35]
[ERROR]  >>        called from: do_companion_libs_for_build[scripts/build/companion_libs.sh@29]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 0:26.00)

camielverdult, 2019-04-03:
I am getting the same issue