pfalcon/esp-open-sdk #360: newlib The FTP download url has expired

qiuwenhui, 2019-04-25:
The FTP download url has expired:
http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0

bulid.log
[DEBUG]    Not at this location: "ftp://sourceware.org/pub/newlib}/newlib-2.0.0"
[DEBUG]    Trying 'http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/qrobot/qrobot/esp-open-sdk/crosstool-NG/.build/tarballs/newlib-2.0.0.tmp-dl' 'http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0' 
[ALL  ]    --2019-04-25 09:44:37--  http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0
[ALL  ]    Resolving mirrors.kernel.org (mirrors.kernel.org)... 149.20.37.36, 2001:4f8:4:6f:0:1994:3:14
[ALL  ]    Connecting to mirrors.kernel.org (mirrors.kernel.org)|149.20.37.36|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2019-04-25 09:44:39--  (try: 2)  http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0
[ALL  ]    Connecting to mirrors.kernel.org (mirrors.kernel.org)|149.20.37.36|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2019-04-25 09:44:41--  (try: 3)  http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0
[ALL  ]    Connecting to mirrors.kernel.org (mirrors.kernel.org)|149.20.37.36|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
[ALL  ]    Giving up.
[ALL  ]    
[DEBUG]    Not at this location: "http://mirrors.kernel.org/sources.redhat.com/newlib/newlib-2.0.0"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_libc_get[scripts/build/libc/newlib.sh@26]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@594]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 0:40.33)
****

qiuwenhui, 2019-04-25:
The solution:
< https://github.com/jcmvbkbc/newlib-xtensa/releases/tag/newlib-2_0_0 >
Download the copied esp-open-sdk/ chrootl-ng /.build/tarballs

bjrobertshen, 2019-07-16:
just download from `ftp://sourceware.org/pub/newlib/newlib-2.2.0.tar.gz` and copied esp-open-sdk/ chrootl-ng /.build/tarballs