pfalcon/esp-open-sdk #362: SDK 3.x - lx106-hal no longer needed??

davydnorris, 2019-04-28:
A few of us have been working extensively with the latest SDK and it appears they have now added entries in the linker script for many, if not all, of the ROM functions previously being built into libhal.

As such we've been able to completely remove libhal from our builds with no side effects, and I was wondering if it is still required in the open-sdk?

pfalcon, 2019-04-29:
The purpose of this project is to provide open-source SDK for esp8266. Is lx106-hal open-source? Yes. Is bootrom open-source? No. The answer to the question posed should be clear from here.

The direction of **this** project is to make source available for more components, not remove components with source available.

closed 2019-04-29 by pfalcon

davydnorris, 2019-04-29:
ok