pfalcon/esp-open-sdk #363: Upgrade esptool.py

iBobik, 2019-05-08:
Currently used version is 3 years old and does not support Python 3. Could we update it?

iBobik, 2019-05-12:
It seems latest master c583756 of esptool works with the rest.

pfalcon, 2019-05-12:
Well, currently used version of esptool.py also works well. Definitely feel free to upgrade it in your setup. The idea behind esp-open-sdk was to provide conservative set of tools known to work (across years).

iBobik, 2019-05-12:
I thing it is too conservative to rely on Python 2.

Jan Pobořil
https://honza.poboril.cz/

pfalcon, 2019-05-12:
Why suddenly?

If you feel so much for Python3, I'd suggest to find a truly worthy project which requires upgrade and help it. My humble suggestion is https://github.com/shedskin/shedskin .

iBobik, 2019-05-12:
It is not option for me (if I understand you correctly) because I need esp-open-sdk for building Micropython and it's build failed with Python 2.

pfalcon, 2019-05-13:
Whose build is failed with Python 2? Both this project and MicroPython is under CI, and there're no known failures.

To some up: As someone who works (and even maintains) a lot of projects, I'm great proponent of proverb "Ain't broken ain't fix it" (because I already spend large part of my life on fixing something, and have a loooong list of what's need to be fixed).

This ticket remains open to let someone show an issue with current esptool.py and propose a plan of validating any new version (and no worries, I'm subscribed to all bugticket updates for esptool, and what I see doesn't make me want to deal with them, if I have a known working version).

alex-eri, 2019-05-25:
Quick fix is using shebang `#!/usr/bin/env python2` instead of `python`

irmen, 2020-02-15:
It think this has become more relevant again now that python 2 is officially no longer supported, and people should have moved to Python 3.5 or newer

projectgus, 2020-02-21:
> people should have moved to Python 3.5 or newer

I haven't tried this, but for anyone who needs Python 3 support it should be possible to remove `bin/esptool.py` from an esp-open-sdk build and `pip install esptool` (or `pip install --user esptool`) to get the latest stable release. The command line arguments are backwards compatible to esptool v1.x.

>  I'm subscribed to all bugticket updates for esptool, and what I see doesn't make me want to deal with them, if I have a known working version

Not really sure what this means, but recent esptool versions are significantly better tested than v1.2 which had no tests at all. The recent versions also have significantly more active users now (Arduino for ESP8266 and ESP32, platform.io, Micropython, ESP-IDF, etc). Suggest this accounts for any perceived excess of issue reports (many of which turn out to be hardware issues presenting as serial port errors - we're the first port of call for most of those.)

pfalcon, 2020-02-21:
> Not really sure what this means,

It means that esptool.py used to be a simple, relatively clean tool, and became a typical "enterprisey" monster. I'm not interested to deal with such in my own free time, in my free time I'm interested to promote cleanliness and simplicity, not support for stuff like serial-over-Internet, which causes issues for some users (abstract example).

All in all, I'll probably add a notice that esp-open-sdk is no longer maintained (which is true) **and** remains a monument of what it was originally intended to be - an example that it's possible to easily build up an open-source toolchain where only warez'ed proprietary toolchains were floating around.

projectgus, 2020-02-25:
> not support for stuff like serial-over-Internet

FWIW: [v1.2](https://github.com/espressif/esptool/releases/v1.2) (the version distributed with esp-open-sdk) is the version which added that support. The only thing which has changed for that feature in more recent versions is noting the limitations of it.

But you are correct that there is more stuff in esptool now compared to back in 2016.

Thanks for all your hard work providing an open source toolchain, when such things were much more difficult to come by.