pfalcon/esp-open-sdk #365: configure: error: could not find bash >= 3.1

Sashkoiv, 2019-07-19:
Ubuntu 18
appears after `make` command
for the solution follow the link
https://bbs.archlinux.org/viewtopic.php?id=244333

vinhjaxt, 2019-08-10:
```
Change line 193 at: esp-open-sdk/crosstool-NG/configure.ac
like this:
` |$EGREP '^GNU bash, version ([0-9.]+)')s`
then run make again
```
[image]

QB4-dev, 2019-08-27:
I can confirm this error on systems with bash version 5+

florosgpl, 2019-10-02:
thanks vin for the heads up.
I hope that pfalcon can fix it in the code, as it is unable to build without editing the config file.
Also, libtool-bin is needed and libexpat-dev is now libexpat1-dev in debian 10+

zenlor, 2019-10-22:
fixed upstream https://github.com/jcmvbkbc/crosstool-NG/pull/56

PL4typus, 2019-11-05:
Can confirm error on Debian 10, can confirm that the fix work

> ```
> Change line 193 at: esp-open-sdk/crosstool-NG/configure.ac
> like this:
>  |$EGREP '^GNU bash, version ([0-9\.]+)')
> then run make again
> ```
> 
> [image]
> p/s: this image is wrong ` ([0-9\.]+)`, not ` ([0-9.]+)`

InSL, 2019-11-14:
@vinhjaxt Thank you! works on raspbian too

tvorogme, 2020-01-30:
Same error on Arch Linux

jcable, 2020-02-02:
I just changed:
                     |$EGREP '^GNU bash, version (3\.[1-9]|4)')
to
                     |$EGREP '^GNU bash, version (3\.[1-9]|4|5)')

Ruzihm, 2020-05-13:
Trying to install this on WSL + ubuntu. jcable's fix worked for me.

sigma-prj, 2020-11-19:
> ```
> Change line 193 at: esp-open-sdk/crosstool-NG/configure.ac
> like this:
>  |$EGREP '^GNU bash, version ([0-9\.]+)')
> then run make again
> ```
> 
> [image]
> p/s: this image is wrong ` ([0-9\.]+)`, not ` ([0-9.]+)`

(3\.[1-9]|4|5|6|7) may be

gretel, 2020-11-26:
now that i figured it's required to use this sdk to build pycopy for esp8266 it's not in usable state 😢

pfalcon, 2020-11-30:
> now that i figured it's required to use this sdk to build pycopy for esp8266 it's not in usable state cry

esp-open-sdk build and then pycopy esp8266 build using it is fine, as exemplified by pycopy CI builds: https://github.com/pfalcon/pycopy/runs/1457821945 .

Maybe it's not fine on some overmodern distros, which is sad, but there's a choice of not falling for overmodern distros, or use docker, etc. When I fall myself for overmodern distros (it happens to everyone eventually, but the keyword is exactly "eventually"), I'll fix it here (can't fix not seeing an issue, right?). Until then, good people above gave hints how to fix it on your side.

gretel, 2020-11-30:
not sure what you are trying to say by "overmodern distros" - i am using `osx` which is said to be pretty common. `micropython` is well supporting this platform so i dont mind closing this instead of fixing an issue reported multiple times here 🤡

pfalcon, 2020-11-30:
> i am using osx

And I'm not using osx, sorry.

gretel, 2020-11-30:
nevermind i get the spirit around here.

hcet14, 2020-12-17:
Same error on Ubuntu 20.04.

ladidadida, 2021-01-23:
I guess a better solution now is described in ticket https://github.com/pfalcon/esp-open-sdk/issues/380

Update crosstools-ng as this solves additional problems.

Pecacheu, 2021-05-13:
> I guess a better solution now is described in ticket #380
> 
> Update crosstools-ng as this solves additional problems.

This almost worked for me on Ubuntu 20.04. I also had to copy over the `local-patches` directory from the old Crosstool-NG.
But now I get this after the script runs for a while:
```
make[2]: Entering directory '/home/ray/esp-open-sdk/crosstool-NG'
make[2]: *** No rule to make target 'xtensa-lx106-elf'.  Stop.
make[2]: Leaving directory '/home/ray/esp-open-sdk/crosstool-NG'
make[1]: *** [../Makefile:135: _toolchain] Error 2
make[1]: Leaving directory '/home/ray/esp-open-sdk/crosstool-NG'
make: *** [Makefile:131: crosstool-NG/.built] Error 2
```

Pecacheu, 2021-05-14:
Alright, think I've finally got it working!

In total here's what I did that finally worked (using Ubuntu 20.04 WSL)
```
sudo apt install <dependencies found in Readme>
git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
rm -rf crosstool-NG
git clone https://github.com/jcmvbkbc/crosstool-NG
cd crosstool-NG
git checkout xtensa-1.22.x
code scripts/build/companion_libs/210-expat.sh
```
Using code, nano, or your editor of choice, add `EXPVERS="2.3.0"` to the start of the file.
Everywhere where it says `${CT_EXPAT_VERSION}`, replace with `${EXPVERS}`
Probably a better way to do this, but this worked.
```
cd ..
build
```
And finished!

*Note: I did also have to tweak some of the dependencies specified in the Readme since some packages weren't found or are installed under a different command (ex python2 rather than python), but nothing too difficult to fix.*

dmascord, 2021-06-22:
@Pecacheu - Version 2.3.0 of Expat is now removed from sourceforge due to exploits on that version, so best to switch to 2.4.1.... more eyes rolling ?

ChrisMacGregor, 2022-01-12:
In addition to the issues described above, I had to fix some other things to get this to build on Ubuntu 21.10.  The end result (which gets me a clean build on Ubuntu 21.10 of an apparently working toolchain) can be found here:

https://github.com/pfalcon/esp-open-sdk/pull/391
https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10
Tarball of built toolchain: https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10

Note that I'm using expat 2.1.0; the comments above suggest that expat 2.4.1 is a drop-in replacement, and (so far) it has not been pulled for security issues.  So you might want to do that differently than I did.

luxkatana, 2024-01-01:
> ```
> Change line 193 at: esp-open-sdk/crosstool-NG/configure.ac
> like this:
>  |$EGREP '^GNU bash, version ([0-9\.]+)')
> then run make again
> ```
> 
> [image] p/s: this image is wrong ` ([0-9\.]+)`, not ` ([0-9.]+)`

This worked for me. Only I switched the
```
|$EGREP '^GNU bash, version ([0-9\.]+)')
```
line 193 with
```
  |$EGREP '^GNU bash, version (5\.[1-9]|4)')
```
Since I am running bash 5+ (to be more specific 5.1.16)

5ch4um1, 2024-04-18:
telling it where to find it in the Makefile worked for me:
`~/esp-open-sdk$ nl Makefile | grep "with-bash"                                                                                                                                     
   124          ./configure --prefix=`pwd` --with-bash=/usr/bin/bash`