pfalcon/esp-open-sdk #367: Build failed in step 'Installing final gcc compiler'

alirezaimi, 2019-10-25:
Hi
I'm getting this error between compiling source :

```
ali@lap:~/git_repo/esp-open-sdk$ make 
Makefile:186: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:181: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
make -C crosstool-NG -f ../Makefile _toolchain
make[1]: Entering directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
../Makefile:186: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:181: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./ct-ng xtensa-lx106-elf
make[2]: Entering directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/ali/git_repo/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20191025.103411
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.10s (at 00:03)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 0.44s (at 00:03)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 49.68s (at 00:53)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 63.83s (at 01:57)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 40.15s (at 02:37)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 50.07s (at 03:27)
[INFO ]  =================================================================
[INFO ]  Installing CLooG for host
[INFO ]  Installing CLooG for host: done in 10.68s (at 03:38)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 21.79s (at 04:00)
[INFO ]  =================================================================
[INFO ]  Installing expat for host
[INFO ]  Installing expat for host: done in 10.05s (at 04:10)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 94.27s (at 05:44)
[INFO ]  =================================================================
[INFO ]  Installing C library headers & start files
[INFO ]  Installing C library headers & start files: done in 0.08s (at 05:44)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C gcc compiler
[INFO ]  Installing pass-2 core C gcc compiler: done in 373.07s (at 11:57)
[INFO ]  =================================================================
[INFO ]  Installing C library
[INFO ]  Installing C library: done in 47.48s (at 12:45)
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[ERROR]    /home/ali/git_repo/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/fold-const.c:9613:1: internal compiler error: Segmentation fault
[ERROR]    make[4]: *** [Makefile:1058: fold-const.o] Error 1
[ERROR]    make[3]: *** [Makefile:3906: all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@545]
[ERROR]  >>        called from: do_gcc_for_host[scripts/build/cc/100-gcc.sh@671]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc.sh@56]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f-dirty/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 15:35.46)
[15:36] / make[2]: *** [ct-ng:152: build] Error 2
make[2]: Leaving directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
make[1]: *** [../Makefile:139: _toolchain] Error 2
make[1]: Leaving directory '/home/ali/git_repo/esp-open-sdk/crosstool-NG'
make: *** [Makefile:131: crosstool-NG/.built] Error 2
```

wi3ik, 2019-11-28:
Did you manage to solve the issue?

jcmvbkbc, 2019-11-28:
This is a host (i.e. x86) compiler crash. Without additional information (e.g. build log) my guess would be that there's not enough memory on the host.

wi3ik, 2019-11-29:
> This is a host (i.e. x86) compiler crash. Without additional information (e.g. build log) my guess would be that there's not enough memory on the host.

Thank you.
Yes, looks like this is the case.
The issue is gone after  I increase available memory on the host.
To be able to execute "make" we should have at least 4Gb memory available.