pfalcon/esp-open-sdk #368: blinky.c gives flash read err, ets_unpack_flash_code

np422, 2019-11-22:
Hello,

I'm new to the esp8266, although not entirely unfamiliar with computers in general, but please don't hesitate to dismiss this issue if there is something fundamental I haven't understood.

I'm using an esp-01 module, the blink sketch from the Arduino 2 IDE works as expected after flashing and resetting so my hw-setup appears to be working.

I've built esp-open-sdk under MacOS 10.12.6.

After make && make flash in the examples/blinky directory the following happens on the serial console after I disconnect GPIO0 from gnd and reset.

```
03:38:23.166 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
03:38:23.203 -> 
03:38:23.203 -> load 0x40100000, len 31364, room 16 
03:38:23.239 -> tail 4
03:38:23.239 -> chksum 0xef
03:38:23.239 -> load 0x88888888, len -2004318072, room 4 
03:38:27.254 -> flash read err, ets_unpack_flash_code
03:38:27.254 -> ets_main.c 
```

I have kept the `static const int pin = 2;` definition in blinky.c and connected GPIO2 to a led/resistor, when using `pinMode(2, OUTPUT);` in setup()  and using `digitalWrite(2, LOW/HIGH);` in the arduino code the led blinks as expected.

I've also manually ran an esptool.py erase_flash before make flash which didn't make any difference.

Is there any special boot-firmware or similar that I need to flash before I can use the esp-open-sdk toolchain?

MrAureliusR, 2020-05-17:
There is indeed a different firmware that needs to be flashed. You can check out my tutorial series on Hackaday which goes into that. Refer to step 5 in this tutorial:
https://hackaday.io/project/160006/instructions
You need to erase the flash, and then flash the ESP8266_NONOS_SDK-2.1.0-18-g61248df/bin/esp_init_data_default.bin file.

CaCTuCaTu4ECKuu, 2023-01-30:
That's not a project error rather than esp itself
When I'm connection ESP-01S to 3.3V I get
```
load 0x40100000, len 1856, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 
```
And if I connect it to 5V (4.8) I see
```
load 0x40100000, len 1856, room 16 
tail 0
chksum 0xef
load 0x88888888, len -2004318072, room 8 
flash read err, ets_unpack_flash_code
ets_main.c 
```
But nevertheless it's not working