pfalcon/esp-open-sdk #369: Build failed in step 'Installing C library'

DasVoelkel, 2019-12-19:
[image]
Compiler crashes. I've had the issue with the Bash parse error and could get around that. Then it crashed with ng because of this error. 
https://github.com/pfalcon/esp-open-sdk/issues/254#issuecomment-298468308
and now it's crashing because of some c-lib and I don't have a clue what to look for now.
Has anyone else had this problem and maybe have a solution ?

jcmvbkbc, 2019-12-19:
> I don't have a clue what to look for now.

Look at the build.log as it suggests.

DasVoelkel, 2019-12-19:
The build log practically has the exact same information.
https://pastebin.com/3zCgzqgx
this has about the same information as the error message i posted, more stuff around it but not really information that helps find what happened

jcmvbkbc, 2019-12-19:
> The build log practically has the exact same information.

It has the following:
```
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memset.S: Assembler messages:
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memset.S:56: Error: unknown opcode or format name 'loopnez'
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memset.S:126: Error: unknown opcode or format name 'loopnez'
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memcpy.S: Assembler messages:
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memcpy.S:68: Error: unknown opcode or format name 'loopnez'
[ALL  ]    /home/zweieuro/Documents/PlatformIO/Projects/master_gate_rtos/esp-open-sdk/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/machine/xtensa/memcpy.S:140: Error: unknown opcode or format name 'loopnez'
```

which suggests that libc doesn't have correct overlay applied to it.

DasVoelkel, 2019-12-19:
.... :/ As someone who is trying this for the first time i don't know what that means...
or how I could fix it

jcmvbkbc, 2019-12-20:
There's a configuration file in the xtensa newlib source tree: newlib/libc/sys/xtensa/include/xtensa/config/core-isa.h
This file specifies which options are available on the target processor. The header is included by xtensa-specific assembly sources so that they only use opcodes present in the target processor.

Normally the header is extracted from the following archive https://github.com/jcmvbkbc/crosstool-NG/blob/37b07f6fbea2e5d23434f7e91614528f839db056/overlays/xtensa_lx106.tar during newlib source extraction process.

You can check the build log to see if that happened.
You can also check the XCHAL_HAVE_LOOPS definition in core-isa.h in the newlib source directory.

DasVoelkel, 2019-12-20:
I am not sure if that helps or how i fix this.
can i just turn this option off? wouldn't that break some functionality if i just remove opcodes from the assembly instruction set?

jcmvbkbc, 2019-12-20:
> I am not sure if that helps

I'm trying to explain what the error that you see means and where it comes from. Sorry if it doesn't help.

> how i fix this.

I usually start fixing things by investigating what went wrong. Knowing how things should work usually helps at that step.
If you could share your complete build.log it would be easier for me to see what happened and suggest further steps.

>wouldn't that break some functionality if i just remove opcodes from the assembly instruction set?

Depends on how you'd do it. It certainly does not seem to be the easiest way to deal with this issue.

> newlib is supposed to be where

You're building the SDK, newlib is built as a part of it, inside crosstool-NG/.build

closed 2020-02-14 by DasVoelkel