pfalcon/esp-open-sdk #370: can't compile on Ubuntu

steeveone, 2019-12-20:
get following error tryng to compile, used following command:
make toolchain esptool libhal `STANDALONE=n

log file error:

[DEBUG]    ==> Executing: 'touch' '/home/steeve/esp/esp-open-sdk/crosstool-NG/.build/src/.ncurses-6.0.patched' 
[DEBUG]    ==> Executing: 'rm' '-f' '/home/steeve/esp/esp-open-sdk/crosstool-NG/.build/src/.ncurses-6.0.patching' 
[DEBUG]    ==> Executing: 'touch' '/home/steeve/esp/esp-open-sdk/crosstool-NG/.build/src/.binutils-2.25.1.extracting' 
[DEBUG]    Entering '/home/steeve/esp/esp-open-sdk/crosstool-NG/.build/src'
[EXTRA]    Extracting 'binutils-2.25.1'
[DEBUG]    ==> Executing: 'mkdir' '-p' 'binutils-2.25.1' 
[DEBUG]    ==> Executing: 'tar' '--strip-components=1' '-C' 'binutils-2.25.1' '-xv' '-f' '-' 
[FILE ]    tar: This does not look like a tar archive
[FILE ]    tar: Skipping to next header
[FILE ]    tar: Exiting with failure status due to previous errors
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Extract[scripts/functions@982]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 1:36.22)

WTPGaming, 2020-05-31:
Same error!

scubasteve1488, 2021-07-20:
Check your build.log file.
`code /esp_open_sdk/crosstool-NG/build.log`

in order to compile crosstool-NG, the make file is using the scripts found 
in this folder: `/opt/esp-open-sdk/crosstool-NG/scripts/build/companion_libs/`
to download dependencies to `/esp-open-sdk/crosstool-NG/.build/tarballs/`

Manually download the required files and copy them into `/esp-open-sdk/crosstool-NG/.build/tarballs/`

run `make`

ChrisMacGregor, 2022-01-12:
Probably not related to the error you describe, but for the benefit of anyone trying to build this on recent Ubuntu (e.g. 21.10) and having trouble:  I had to collect fixes from various places, plus my own fix for a compile error in gcc-4.8.5.  The end result (which gets me a clean build on Ubuntu 21.10 of an apparently working toolchain) can be found here:

https://github.com/pfalcon/esp-open-sdk/pull/391
https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10
Tarball of built toolchain: https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10

f1ac0, 2022-10-08:
> Probably not related to the error you describe, but for the benefit of anyone trying to build this on recent Ubuntu (e.g. 21.10) and having trouble: I had to collect fixes from various places, plus [...]

Thank you Sir for your patching work, your repo, and your message here that pointed it out !

I am running linux Mint 21 which is based on Ubunty Jammy. In addition to your repo, because of the python replacement, I had to:
- install the python3-is-python-dev package so that the build will finish successfully
- replace the esptool directory with https://github.com/espressif/esptool, in the root of the esp-open-sdk
- install python3-serial instead of python-serial
Hope this helps !

I haven't tried the other forks yet since I am building an old application.