pfalcon/esp-open-sdk #371: hi! when executing a command: Make ---> Makefile:74: recipe execution error for the target «image.elf.bin» (Ubuntu 16.04 x32, Python 2.7.12, Nodemcu esp8266)

kasimusNo, 2020-01-21:
:~/q/esp-open-sdk/examples/espthernet$ make
~/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -mlongcalls -I ~/esp-open-sdk/esp_iot_sdk_v1.5.2/include -Imyclib -Iinclude -Iuser -Os -I ~/esp-open-sdk/esp_iot_sdk_v1.5.2/include/ -Icommon -DICACHE_FLASH -Ietherhelp	-Wl,--gc-sections -flto driver/uart.c common/http.c common/mystuff.c common/flash_rewriter.c common/commonservices.c common/http_custom.c common/mdns.c common/mfs.c user/custom_commands.c user/i2sduplex.c user/user_main.c user/manchestrate.c etherhelp/net_compat.c etherhelp/crc32.c etherhelp/tcp.c etherhelp/iparpetc.c  -nostdlib -L~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib -L~/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/libgcc.a -g ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libmain.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libpp.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libnet80211.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libwpa.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/liblwip.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libssl.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libupgrade.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libnet80211.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/liblwip.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libphy.a ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib/libcrypto.a ~/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/libgcc.a -T ~/esp-open-sdk/esp_iot_sdk_v1.5.2/ld/eagle.app.v6.ld -B ~/esp-open-sdk/esp_iot_sdk_v1.5.2/lib -o image.elf.bin
/bin/sh: 1: /home/sd/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found
Makefile:74: recipe execution error for the target «image.elf.bin»
make: *** [image.elf.bin] Error 127

pfalcon, 2020-01-21:
Cannot reproduce. See https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

closed 2020-01-21 by pfalcon

kasimusNo, 2020-01-21:
Sorry! I just started to study this area! namely, work with nodemcu esp8266 and wanted to upload espthernet to this device (https://github.com/cnlohr/espthernet) installed esp-open-sdk according to instructions (https://github.com/pfalcon/esp-open-sdk)! there is no problem with the blink example! and with this, this error pops up during compilation!