pfalcon/esp-open-sdk #375: gcc: error: unrecognized command line option '-mlongcalls'

vyorkin, 2020-04-08:
The build fails with
```
gcc: error: unrecognized command line option '-mlongcalls'
```

In `config.log` I have the following lines:
```sh
configure:2884: $? = 0
configure:2873: gcc -v >&5
Using built-in specs.
COLLECT_GCC=/nix/store/hmzcr79qmdzhk2b66nk3zifgkcx5cm1n-gcc-9.2.0/bin/gcc
COLLECT_LTO_WRAPPER=/nix/store/hmzcr79qmdzhk2b66nk3zifgkcx5cm1n-gcc-9.2.0/libexec/gcc/x86_64-unknown-linux-gnu/9.2.0/lto-wrapper
Target: x86_64-unknown-linux-gnu
Configured with: 
Thread model: posix
gcc version 9.2.0 (GCC) 
configure:2884: $? = 0
configure:2873: gcc -V >&5
gcc: error: unrecognized command line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:2884: $? = 1
configure:2873: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.
configure:2884: $? = 1
configure:2904: checking whether the C compiler works
configure:2926: gcc -mlongcalls  -nostdlib conftest.c  >&5
gcc: error: unrecognized command line option '-mlongcalls'
configure:2930: $? = 1
configure:2968: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "lx106-hal"
| #define PACKAGE_TARNAME "lx106-hal"
| #define PACKAGE_VERSION "rc-2010.1"
| #define PACKAGE_STRING "lx106-hal rc-2010.1"
| #define PACKAGE_BUGREPORT ""
| #define PACKAGE_URL ""
| #define PACKAGE "lx106-hal"
| #define VERSION "rc-2010.1"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:2973: error: in `/home/vyorkin/projects/github/esp/esp-open-sdk/lx106-hal':
configure:2975: error: C compiler cannot create executables
See `config.log' for more details
```

pfalcon, 2020-04-08:
I just re-run a Travis CI build, which is green: https://travis-ci.org/github/pfalcon/esp-open-sdk/builds/672660589 . You can see how the build log differs from your results, and think what to do about (the easiest solution might using known-working OS in a container or similar).

vyorkin, 2020-04-08:
Thanks. I've got it working by "manually" building the `lx106-hal` as described [here](https://github.com/tommie/lx106-hal#hardware-abstraction-library-for-xtensa-lx106). Just tried compiling and uploading the "blinky" example

closed 2020-04-08 by vyorkin