pfalcon/esp-open-sdk #377: Cannot compile, could not find curses header

beingPurple, 2020-07-09:
I am getting an error when trying to compile: 
```
configure: error: could not find curses header, required for the kconfig frontends
make[1]: *** [../Makefile:149: _ct-ng] Error 1
make[1]: Leaving directory '/home/user/esp-open-sdk/crosstool-NG'
make: *** [Makefile:145: crosstool-NG/ct-ng] Error 2
```
Any help is appreciated! This is my first time posting an issue to an open source project like this, so I apologize in advance if I am out of line.

jcmvbkbc, 2020-07-09:
Have you followed [this part](https://github.com/pfalcon/esp-open-sdk#requirements-and-dependencies) of the instruction, specifically do you have ncurses development package installed?

beingPurple, 2020-07-09:
That was it! It was listed in my repository as ncurses-devel rather than ncurses-dev. Thank you for the help!