pfalcon/esp-open-sdk #379: failed to build it in MacOS

kenjichanhkg, 2020-08-19:
when I followed the instruction, the first command failed.....
[image]

riazbapoo, 2020-10-15:
> when I followed the instruction, the first command failed.....
> [image]

same here. have you found a solution?

gretel, 2020-11-26:
samesame.

kenjichanhkg, 2020-11-28:
still can't

mnbpdx, 2021-03-08:
I've got the solution to this particular issue, though I think there's still problems down the road on MacOS. I'm having trouble getting `make` to work.

As your error says, homebrew/dupes has been deprecated. I don't know much about what's going on, but its explained [here](
https://stackoverflow.com/questions/45124717/whats-the-migration-from-homebrew-dupes-to-homebrew-core).

Just add the dependencies like this:

`brew install <dependency>`

Side note, brew install didn't seem to recognize the `--with-default-names` flag, so I omitted it. Hopefully that didn't break anything.

mnbpdx, 2021-03-08:
Right now I'm stuck on this error, hit me up if you get past it:

`
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
`

gretel, 2021-03-08:
this project looks.. done to me. move on folks 😄

mnbpdx, 2021-03-08:
More hot tips: you'll have to make sure you have `gawk` not `awk` installed, or you'll get  a `make` error:

`brew install gawk`

Had a similar error for `objcopy`/`gobjcopy`. Fixed it by adding binutils stuff to my path per #350.

`export PATH="/usr/local/opt/binutils/bin:$PATH"`

mnbpdx, 2021-03-08:
@gretel Got a good alternative then? 

I'm new to open source, haven't figured out when to give up and move on

gretel, 2021-03-08:
> @gretel Got a good alternative then?

the last commit to this repository was on 27 Nov 2018. the original author has not taken care of 109 open issues and 25 pull requests in the meantime.

due to the targets being commercial (not open source) hardware there is some controversy intrinsic for the whole of this.

> I'm new to open source, haven't figured out when to give up and move on

even if you would get the toolchain running you are still limited to old versions of the dependencies required and now-outdated hardware targets.

it's up to you to decide if you want to stick with idealism from the past or head on to the now-current reality: `espressif` got super sucessful and is still not providing a truly open (source) hardware design.

personally, i do accept the commercial nature of mass-produced (cheap) microcontrollers and develop using [micropython](https://github.com/micropython/micropython).

mnbpdx, 2021-03-09:
Thanks for your thorough response, @gretel, I'll go on a dive with all those keywords and see where it gets me.

gretel, 2021-03-09:
> Thanks for your thorough response, @gretel, I'll go on a dive with all those keywords and see where it gets me.

i'd say wifi repeaters are not a good idea to run in most cases. either you have issues with the access point you are using and/or a better off using multiple inter-linked access points instead. maybe [openwrt](https://openwrt.org/about) is worth a look.