pfalcon/esp-open-sdk #380: crosstools-NG/configure.ac

dbetz, 2020-09-04:
esp-open-sdk fails to build on recent versions of Ubuntu that include bash 5.0.17(1)-release. This is because the configure.ac script for the included version of crosstools-NG only recognizes versions of bash up to 4. Please update crosstools-NG to allow more recent versions of bash.

Thanks for your work on this project. It's very helpful.

drawkula, 2020-09-04:
Try...
```
(yeti@kumari:6)/opt/esp-open-sdk$ cd crosstool-NG/
(yeti@kumari:6)/opt/esp-open-sdk/crosstool-NG$ git diff
diff --git a/configure.ac b/configure.ac
index 5d512fe8..872a73de 100644
--- a/configure.ac
+++ b/configure.ac
@@ -190,7 +190,7 @@ AC_CACHE_VAL([ac_cv_path__BASH],
 AC_CACHE_CHECK([for bash >= 3.1], [ac_cv_path__BASH],
     [AC_PATH_PROGS_FEATURE_CHECK([_BASH], [bash],
         [[_BASH_ver=$($ac_path__BASH --version 2>&1 \
-                     |$EGREP '^GNU bash, version (3\.[1-9]|4)')
+                     |$EGREP '^GNU bash, version (3\.[1-9]|4|5)')
           test -n "$_BASH_ver" && ac_cv_path__BASH=$ac_path__BASH ac_path__BASH_found=:]],
         [AC_MSG_RESULT([no])
          AC_MSG_ERROR([could not find bash >= 3.1])])])
```
...™worked for me™ on Debian10 a while ago.  Reubilding now to check it...

...ok...
```
          AC_MSG_ERROR([could not find bash >= 3.1])])])
```
...needs an update too... somewhen... mayhaps... ॐmmmmmm... ;-)

_o/"

dbetz, 2020-09-04:
Yes, I did that. I just thought it would be helpful if that change was made in the repository.

pfalcon, 2020-09-06:
Thanks for the report. It's a matter of fact that I don't maintain esp-open-sdk anymore, and it stays in a static, known-working (with the tools of its generation) shape. So, I unlikely will update this repo to a newer crosstools-NG version, as then someone would need to re-triage "known working" status, and that's not easy.

(I do however still rename tickets so their titles corresponded to an issue reported and other users had a chance to discover them ;-) ).

suculent, 2020-10-29:
Just stumbled upon this issue when updating my docker builder.

Actually, the source repository of crosstool-ng has changed and the new one is maintaine: While building the dockerized toolchain (maybe the case-sensitive rename is not necessary, nevertheless it works), i've just replaced the original source with new repo as a patch:

```
cd /home/nodemcu/esp-open-sdk/ && \
rm -rf ./crosstool-NG && \
git clone https://github.com/crosstool-ng/crosstool-ng && \
mv ./crosstool-ng ./crosstool-NG
```

Haox1n, 2020-11-26:
> Just stumbled upon this issue when updating my docker builder.
>
> Actually, the source repository of crosstool-ng has changed and the new one is maintaine: While building the dockerized toolchain [...]

thanks for your msg here.
it helped me (as a noob) out.
thx again
Stay well

Pecacheu, 2021-05-13:
> Thanks for the report. It's a matter of fact that I don't maintain esp-open-sdk anymore, and it stays in a static, known-working (with the tools of its generation) shape. So, I unlikely will update [...]

I don't know about 'known working' when as I just discovered apparently there are some dependencies it tries to download off the web that literally don't exist anymore at that URL, which means build will probably fail for ALL users. (#365)

ChrisMacGregor, 2022-01-12:
A possibly-simpler fix for the bash-version-detection issue is this:

https://github.com/ChrisMacGregor/esp-open-sdk/commit/3fc1930616fb6ffeecefe30f71857b1547a578bb#diff-76ed074a9305c04054cdebb9e9aad2d818052b07091de1f20cad0bbac34ffb52

I subsequently ran into a couple of other issues that needed patches.  The end result (which gets me a clean build on Ubuntu 21.10 of an apparently working toolchain) can be found here:

https://github.com/pfalcon/esp-open-sdk/pull/391
https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10
Tarball of built toolchain: https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10