pfalcon/esp-open-sdk #384: Make error: Expat-2.1.0 renamed on sourceforge

emailbsuv, 2021-06-20:


scubasteve1488, 2021-07-20:
I had no luck with modifying config files to adapt to rename.
Easiest solution I've found. 

Download "expat2.3.0-RENAMED-VULNERABLE-PLEASE-USE-2.4.1-INSTEAD.tar.gz" 
from: https://sourceforge.net/projects/expat/files/expat/

rename to expat2.3.0.tar.gz and move to /esp-open-sdk/crosstool-NG/.build/tarballs/

cd /esp_open_sdk/
make

EtchedPixels, 2021-08-05:
Not very smart - expat 2.3 has serious security holes and was deliberately moved. Your build is now totally vulnerable to attack. The better fix is to use expat 2.4.1 or later like it tells you to.

derbroti, 2021-08-11:
For completeness: 

- edit crosstool-NG/config/companion_libs/expat.in

- replace all occurrences of 2.1.0 with 2.4.1
(This includes the underscore ones like: "EXPAT_V_2_4_1")

- run make and everything will be fine