pfalcon/esp-open-sdk #386: i can't install isl 0.14

curla92, 2021-10-05:
i tried to put manual the file into tarballs folder but when i try to compile i have an error.
This error says that missing the configure file of ISL 0.14.

How can i solve this ? thank y

glenn20, 2021-10-20:
Not directly answering the question, but, I used the following patch to correct the download url for isl 0.14:
```
--- a/scripts/build/companion_libs/121-isl.sh
+++ b/scripts/build/companion_libs/121-isl.sh
@@ -14,7 +14,8 @@ if [ "${CT_ISL}" = "y" ]; then
 # Download ISL
 do_isl_get() {
     CT_GetFile "isl-${CT_ISL_VERSION}" \
-        http://isl.gforge.inria.fr
+        ftp://gcc.gnu.org/pub/gcc/infrastructure
+        # http://isl.gforge.inria.fr
 }
 
 # Extract ISL
```
I hope that's helpful. Worked for me, but I'm not sure what the canonical source should be or why the manual method failed.

ChrisMacGregor, 2022-01-06:
I had to also make the change here:
      crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6f/scripts/build/companion_libs/121-isl.sh

and then a similar change in 210-expat.sh in both locations (crosstool-NG/scripts/build/companion_libs/210-expat.sh and
crosstool-NG/lib/crosstool-ng-1.22.0-60-g37b07f6f/scripts/build/companion_libs/210-expat.sh), replacing:

      http://downloads.sourceforge.net/project/expat/expat/${CT_EXPAT_VERSION}
with:
      https://src.fedoraproject.org/repo/pkgs/expat/expat-${CT_EXPAT_VERSION}.tar.gz/dd7dab7a5fea97d2a6a43f511449b7cd

because expat-2.1.0.tar.gz on sourceforge had been renamed to "expat-2.1.0-RENAMED-VULNERABLE-PLEASE-USE-2.3.0-INSTEAD.tar.gz"...  I considered seeing if it would work using 2.3.0 as suggested, but 2.3.0 similarly redirects to 2.4.1, and for the moment I'm not concerned about it.  This should probably be fixed, but it looks like no one has touched these tools for years now.

PsuFan, 2022-01-28:
I cant get isl 0.14 or expat-2.1.0. Patches didn't work for me. Everyone's "updated" "maintained" forks didn't work for me either. They don't even bother to update the readme. Download manually from http://mirror.opencompute.org/onie/crosstool-NG/ and place in esp-open-sdk/crosstool-NG/.build/tarballs

I WISH SOMEONE WOULD MAINTAIN ACTUAL INSTRUCTIONS ON HOW TO INSTALL MICROPYTHON FOR ESP8266 INCLUDING ALL COMMANDS, INCLUDING GIT, MODIFICATIONS TO $PATH ETC. I was screwed for a while with $PATH because I used ~ which worked fine for me but not compiler...

igagis, 2022-01-28:
@PsuFan see https://github.com/micropython/micropython/issues/8163 , it is an open issue

ChrisMacGregor, 2022-01-28:
> I cant get isl 0.14 or expat-2.1.0. Patches didn't work for me. Everyone's "updated" "maintained" forks didn't work for me either. They don't even bother to update the readme. Download manually from [...]

@PsuFan Does https://github.com/ChrisMacGregor/esp-open-sdk work for you?

Fair point about the readme files.  I didn't because I was trying to keep my repo as close as possible to the original, but there's probably no harm in updating the readme.