pfalcon/esp-open-sdk #387: ANY MAINTAINED FORK?

igagis, 2021-12-01:
As far as I know @pfalcon has abandoned the project, so this repo is no longer updated/maintained. For example, see https://github.com/pfalcon/esp-open-sdk/pull/344#issuecomment-437567327 comment on one of the open PRs.

Are there any forks of this which are maintained?
Especially, those which have updated to latest SDK (3.0.x) and have #385 fixed etc.

If there is no well known maintained fork, maybe it is time to create one?

someburner, 2021-12-01:
Mine is somewhat maintained: https://github.com/someburner/esp-open-sdk

I do not use SDK3.x though due to wifi stability regressions and incompatibility with various hacks the Arduino project has implemented, see [here](https://github.com/esp8266/Arduino/issues/5784). It uses [nonos lwip2 hack](https://github.com/d-a-v/esp82xx-nonos-linklayer) and GCC7. I think with the crosstool-ng version I'm on there is one broken download link you may have to manually download the tarball and place it in the crosstool build downloads folder.

igagis, 2021-12-01:
@someburner 
Well, I'm not using Arduino, so I moved to SDK 3.x as soon as it was released. So, your fork wouldn't work for me, unfortunately.

AFAIK, current version of NONEOS SDK is 3.0.5: https://github.com/espressif/ESP8266_NONOS_SDK/releases
So, I'd like to have that.

NONEOS SDK is considered EOL, so no new features will be added, but bug fixes - yes. So, I consider using the recent version of the SDK should not harm, and API should not change drastically between 3.x versions. For example between version 2.x and 3.x they introduced memory partitions which broke SDK's backwards compatibility, i.e. users had to update their code to use partitions to be able to build with SDK 3.x.

someburner, 2021-12-01:
I'm not using Arduino either, but have utilized some of their lower-level SDK hacks. They're not strictly necessary though. I have updated between 2.x -> 3.x -> 2.x without issue, but I use rboot so I could see that breaking for other versions.

It's actually pretty easy to update to 3.x and if you look at the insights -> network tab of this repo there are quite a few forks that have done so, maybe [this one](https://github.com/IoTAqua/esp-cross-sdk/commit/db3bb728b7231f2de8f9bcfba588f0687090527d)?

The main issue is that crosstool-ng needs to be kept updated as download links break, but also later crosstool versions have gotten rid of older GCC versions and you need to add patches in there for newer GCC versions. Looking at my commit history the only thing I had to do in this SDK repo for 3.x was make a [new patch for ctypes.h](https://github.com/someburner/esp-open-sdk/commit/f98ee40b8133f6ff8de6b8b9cec86e1d4a589745). Not sure if that is still all the is required now.

igagis, 2021-12-02:
@someburner so, if I fork your repo, what do I need to do to update to SDK v3?

igagis, 2021-12-02:
Ok, I have just made a fork of pfalcon's repo. The 600th fork :).

https://github.com/igagis/esp-open-sdk

I'm going to update it to latest SDK and fix building problem. I also plan to maintain that fork and accept PRs which would keep the components up to date.

Also, updating to some later GCC version would not harm.

uzi18, 2021-12-02:
@someburner has got tree for v3.x maybe you can use it as starting point?

igagis, 2021-12-02:
@uzi18 good idea!

igagis, 2021-12-02:
@someburner if I checkout `3.x` branch from your repo, will it be the SDK 3.x?

igagis, 2021-12-04:
Ok, my fork now builds and is updated to sdk 3.0.0

ChrisMacGregor, 2022-01-12:
Alternately, a fork with the minimal set of changes to get a clean build on Ubuntu 21.10 of an apparently working toolchain can be found here:

https://github.com/pfalcon/esp-open-sdk/pull/391
https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10
Tarball of built toolchain: https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10

A maintained fork with more-current versions of things, such as the one described above, may be a better option.  I'm offering a minimally-tweaked option for those who prefer that, or for comparison purposes if for some reason the newer tools don't work for you, or whatever.  If your code doesn't work with either the updated toolchain or the minimally-tweaked one, then the problem is more likely in your code, rather than the toolchain...

igagis, 2022-12-01:
fork abandoned

closed 2022-12-01 by igagis