pfalcon/esp-open-sdk #393: PLEASE UPDATE INSTRUCTIONS FOR PYTHON3

Jibun-no-Kage, 2022-08-14:
The following will not work for Python3 based Debian 11...

$ sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
    sed git unzip bash help2man wget bzip2

eku, 2022-08-14:
Why don't you provide a PR that changes it?

Jibun-no-Kage, 2022-08-14:
Actually last night I worked on that, but was unsuccessful.  Could not get the tool-chain to complete.  Ran into 2 or 3 issues, that are similar to what others have posted trying to get a compile environment up and running.  So clearly to update anything takes some deeper knowledge than I have at the moment.  I have zero experience with Linux GCC environment.

eku, 2022-08-16:
If you took some time an read through older issue, you would have noticed, that this project might no longer be maintained an there are forks, that address all the mentioned issues. Did you try this [fork](https://github.com/thierer/esp-open-sdk)?

Jibun-no-Kage, 2022-08-16:
So you start the comment with an insult?  If you took some time.... I have spent about 16 hours trying to figure out how to get any variant of any google example of the tool-chain and SDK environment to compile and work correctly.  And no, I have not exhausted everything possible option.  No wonder people state that GitHub is getting toxic.

andrei-gavrila, 2024-01-08:
The issue https://github.com/pfalcon/esp-open-sdk/issues/396 contains all the required steps to build on the latest versions of all :)